frangoteam/FUXA #1524: [BUG]Configuring the home screen based on the logged in user

robsori, 2024-12-05:
**Describe the bug**
I tried to open a View based on the logged in user but after a short while I am directed to the MainView configured in Layout Settings.

**To Reproduce**
Create two users, user1 and user2 and configure a home screen for each. Then log in with each user in turn.

**Expected behavior**
I expect that if I log in with user "user1" I will stay on the "View1" page or if I log in with user "user2" I will stay on the "View2" page.

**Screenshots**
[image]

**Environment**
 - FUXA version: v1.2.1
 - Node.js version: 18.20.4
 - npm version: 6.14.18
 - Platform/OS: Windows 10 Pro
 - Browser: Any

unocelli, 2024-12-06:
Hi, if you do not enable the authentication, it cannot work. how do you know which user has logged on?

unocelli, 2024-12-11:
I’m going to close this as resolved. let me know if you have any issues.