frangoteam/FUXA #1525: Hello, what is the oldest version of Firefox that Fuxa supports? I'm currently encountering an issue where it cannot open on Firefox and gets stuck on the homepage logo page

fengchenyu, 2024-12-05:
error:syntaxError: expected property name.got '{'
Syntaxtrror, expected expression, got '?'
[image]
Syntaxtrror expected expression,at '?'

**Describe the bug**
A clear and concise description of what the bug is.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Environment**
 - FUXA version: 
 - Node.js version: 
 - npm version:
 - Platform/OS:
 - Browser:

unocelli, 2025-03-04:
Hi, will be better to explain with which version don't work...with the last is working as expected

unocelli, 2025-05-04:
Closing this due to inactivity. If the issue is still relevant or unresolved, feel free to reopen.

closed 2025-05-04 by unocelli