frangoteam/FUXA #1526: SSL eMail port 465

robbudge, 2024-12-05:
**Describe the bug**
   With email server port set to 465, sendMail returns 'Greeting never received'
in file /server/runtime/notificator/index.js
initially I hardcoded secure:true,
emails sent without problems 

I then modified 
secure: (smtpServer.port === 465) ? true : false, // true for 465, false for other ports 
to 
secure:((smtpServer.port == 465) ? true : false), // true for 465, false for other ports
and emails continued to work correctly.

No idea why or how, but secure:true worked for both my SSL email servers, and the modified code appears to be working.

could just be my servers, but I have never been able to send email via any of our SSL servers.

henjoe, 2024-12-06:
What smtp server you use?  gmail or customized?

robbudge, 2024-12-06:
We have VPS Servers running separate domains, Fully Qualified with SSL certificates.
basically your standard main stream corporate email servers.

unocelli, 2025-03-04:
I assume the issue has been resolved in the meanwhile. I'm therefore going to close this issue.

closed 2025-03-04 by unocelli