frangoteam/FUXA #1529: [Question] Save project does not reflect changes on "save project as" file location.

OptifySudarshanPatil, 2024-12-07:
It's has been  joy of using FUXA as go to dashboard maker.
I'm curious, where or at which location file get save after "Save project" button pressed. 
I can use "Save project as", but cant imagine what is "default" project save location which reflects changes done.(`it supposed to be overwrite "save project as" file location.`)

[image]

## OS: Windows 10 Pro 64x

## Fuxa Version 1.2.1-1988

[image]

unocelli, 2024-12-18:
Hi, in [wiki](https://github.com/frangoteam/FUXA/wiki/HowTo-save-load-Project) you have little description

OptifySudarshanPatil, 2024-12-25:
> Hi, in [wiki](https://github.com/frangoteam/FUXA/wiki/HowTo-save-load-Project) you have little description

Thanks!

closed 2024-12-25 by OptifySudarshanPatil