frangoteam/FUXA #1533: Question: how to use Read Scale Script

jonbur, 2024-12-09:
Hi, I hope you can help. I am having trouble with Read Scale Script and can't find any documentation.

I have several tags coming from mqtt that hves the value either 'on' or 'off', as below:

[image]

I would like to use this to drive Actions on a shape. I understand that Actions can only be driven by numerical values and therefore I am looking to convert the text values to equivalent numerical/binary values (1/0). I have created the script below and assigned it as the Read Scale Script.

[image]

[image]

However, this does not appear to work. There is no error in the logs. Even if I change the script to simply just 'return 1' in all cases I don't see a change to the value of the tag. 

Am I missing something?

Thank you very much.

MatthewReed303, 2024-12-09:
@jonbur at the bottom of the page https://github.com/frangoteam/FUXA/wiki/HowTo-Devices-and-Tags

jonbur, 2024-12-09:
Hi,

Thanks for this. I have seen this and believe I have followed the design pattern shown, but it does not look like the script is being applied / changing the output. 

Just to make sure I have this correct conceptually, the 'value' that is passed into the read script is the tag value that is read from the integration connector (in this case mqtt) and then the script manipulates it and whatever is returned should then be what we see when that tag is used elsewhere?

Thank you

henjoe, 2024-12-15:
Your script is actually correct. I dont know why it is not working.

unocelli, 2024-12-15:
Hi, try to debug with a console.log(value);

jonbur, 2024-12-15:
Thanks. If I run the script in the script editor with console logging, I get the below:

[image]

This seems fine (although I note I don't get return '0' when I call with 'off'...

Is there a logger object I can use to log to fuxa.log from within the script so I can see whether this script is actually getting caleld by the 'read scale script' defined on the tag?

Thanks

henjoe, 2024-12-15:
Hi @unocelli  the  issue might be related to #1435

unocelli, 2024-12-18:
Try to return a string '0'

jonbur, 2024-12-18:
Hi,

Sorry, this still does not appear to work.

With the script below:

[image]

Applied as the 'Read Scale Script' on a tag as per the below:

[image]

The value does not change (see circled in red). It also does not change in the action on the object in the editor - it is still 'on' or 'off'.

Even is I just do this:

[image]

the value does not change.
It is like the script is not being called at all...

Thanks

jonbur, 2024-12-19:
Thanks very much, I’ll test the latest version over the weekend :-)

851666395, 2024-12-20:
> Thanks very much, I’ll test the latest version over the weekend :-)

I have also encountered similar problems in webAPI。

[image]

[image]

851666395, 2024-12-20:
> Hi, sorry, you are right! in mqtt is missing a reference to process the script

HI,I have also encountered similar problems in webAPI。

unocelli, 2024-12-20:
@851666395 Hi, Yes should be fixed in mqtt, ethernetip and httprequest (WebApi)

851666395, 2024-12-21:
> @851666395 Hi, Yes should be fixed in mqtt, ethernetip and httprequest (WebApi)

thank you，I'll try again when it's fixed。

jonbur, 2024-12-21:
Thanks, I can confirm the latest release is working for me.
Thank you.

unocelli, 2025-01-01:
I’m going to close this as resolved. let me know if you have any issues.

closed 2025-01-01 by unocelli

H-hejin, 2025-06-27:
> I’m going to close this as resolved. let me know if you have any issues.

Why do the Pro version I downloaded still have the issues mentioned above, which makes me very distressed? 1.2.0

unocelli, 2025-07-09:
You’re using an old version, please download the latest one (currently 1.2.5), where those issues should be fixed.