frangoteam/FUXA #1536: [Question]WAR]  try to create s7300 but plugin is missing!

otli-max, 2024-12-12:
Node-snap7 was installed already,  there are still warning" [WAR]  try to create s7300 but plugin is missing!" in the terminal, could you please tell me is there anything i missed? There are some screen snap below:
[image]
[image]
[image]

unocelli, 2024-12-19:
Hi, try to install it only in fuxa/server

unocelli, 2025-03-04:
I assume the issue has been resolved in the meanwhile. I'm therefore going to close this issue.