frangoteam/FUXA #1546: [QUESTION] How to add Tags from the WebAPI

henjoe, 2024-12-22:
**Describe the Question**
I have a web api server running on HTTP://localhost:5050/api/test

And this return a dummy data: 
```
[
    {
        id: 'tempA',     // UUID
        name: 'tempA',
        value: 123,
        type: 'number'  // number|boolean|string,`
    },
    {
        id: 'tempB',    // UUID
        name: 'tempB',
        value: 23.5,
        type: 'number'  // number|boolean|string,`
    }
]
```
When adding a device with type WebAPI, I can successfully received this data.

[image]

Now my question is how can I add the Tags from the WebAPI device? when I tried to browse the Tags it gives me something like this: 
[image]

The issue is I cannot select the Tag to add on my Device Taglist. 
Am I doing something wrong here?

unocelli, 2024-12-22:
It is an array, select the ID and Value
[image]

henjoe, 2024-12-23:
Hi @unocelli  thanks for clarification, I will try this. I am thingking that it should have a checkbox (like on the opcua). 

Regards,

closed 2024-12-25 by henjoe