frangoteam/FUXA #1547: [FEATURE] When using the view, I noticed that the vertical scrolling area is quite large. Is it possible to limit the scrollable range?

KWYHDHJ, 2024-12-23:
**Describe the feature**
I often accidentally scroll with the mouse wheel while using it, which results in only being able to see a small portion of the view.

**Describe the solution you'd like**
Is it possible to restrict the range of the negative ruler in the view?

**Additional context**
[image]
[image]

KWYHDHJ, 2024-12-23:
Is the view editing a method you encapsulated yourself? I tried modifying the corresponding code in the client folder but couldn't find where the settings are located.

unocelli, 2024-12-23:
Hi, you can zoom In to reduce it. 
Shift + SCROLLER: zoom by mouse position

KWYHDHJ, 2024-12-24:
> Hi, you can zoom In to reduce it. Shift + SCROLLER: zoom by mouse position

Yes, I know that the view size can be zoomed by using Shift + SCROLLER: zoom by mouse position, but wouldn't it be better to add some restrictions outside the view editing area, apart from the white content? After all, we don't use the parts outside the view editing area, so adding some restrictions could optimize the user experience.