frangoteam/FUXA #1549: [BUG] High CPU Usage When Viewing “Edit Device Tags” Page when working with many tags

eduardoexact, 2024-12-23:
I found the following bug when handling large number of tags. I observed that the problem just appear when the user open the "Edit Device Tags". It seems that fuxa is pulling all the values even when just a few are beign displayed (10, 25 or 100 depending on the pagination). Solving this can help to handle a large number of tags.

Describe the bug
When the “Edit Device Tags” page is opened and a large number of tags (e.g., hundreds or thousands) are loaded, the CPU usage spikes significantly. This appears to be caused by the application requesting and processing all device tags at once, resulting in a high load on the server.

To Reproduce

Go to the application’s “Devices” section.
Click on the “Edit Device Tags” icon for a device that contains many tags.
Observe that the UI attempts to load the entire list of tags in one go.
Notice a substantial increase in CPU usage on the server or in the Node.js process.

Expected behavior

Only the tags the user needs to see (e.g., a page of 25 items) should be requested and displayed at a time, preventing excessive CPU consumption. Scrolling or paginating should fetch subsequent portions of the tag list without taxing the system.

What I think might be happening:

When users open the “Edit Device Tags” page, the frontend requests *all* MQTT tags at once. This leads to high CPU usage on the server (Node.js) because:
1. The server fetches and processes *all* tags every time.
2. The client receives a very large payload (all tags) even if only a subset are shown in the UI.

From profiling, we’ve seen the CPU usage spike in `updateDeviceValues()` and other functions that handle large arrays or repeated filters. This happens especially when there are many active MQTT tags or many connected clients.

**Desired Outcome**  
- Only fetch and display the tags that are actually needed (for example, 25 per page, 10 or 100 as in the table list), which significantly reduces CPU usage and payload size.
- Implement pagination (or partial fetch) so the server and client handle less data at once.

## Proposed Solution
1. **Frontend** (`client/src/app/_services/hmi.service.ts`):
   - Change `askDeviceTags(deviceId: string)` to include pagination parameters:
     ```ts
     public askDeviceTags(deviceId: string, page: number = 1, pageSize: number = 25) {
       if (this.socket) {
         this.socket.emit(IoEventTypes.DEVICE_TAGS_REQUEST, {
           deviceId,
           page,
           pageSize
         });
       }
     }
     ```
   - Update the component that calls `askDeviceTags(...)` to include pagination logic (e.g., “Next Page,” “Previous Page,” etc.).

2. **Server** (`runtime/index.js` or equivalent):
   - In the `socket.on(DEVICE_TAGS_REQUEST, ...)` handler, parse `page` and `pageSize`, slice the array of all device tags, and return only the requested portion:
     ```js
     socket.on(Events.IoEventTypes.DEVICE_TAGS_REQUEST, async (message) => {
       try {
         const deviceId = message.deviceId;
         const page = message.page || 1;
         const pageSize = message.pageSize || 25;

         const allTags = await devices.getDeviceTagsResult(deviceId);
         const start = (page - 1) * pageSize;
         const end = start + pageSize;
         const pageTags = allTags.slice(start, end);

         message.result = {
           page,
           pageSize,
           total: allTags.length,
           tags: pageTags
         };

         io.emit(Events.IoEventTypes.DEVICE_TAGS_REQUEST, message);
       } catch (err) {
         message.error = err.toString();
         io.emit(Events.IoEventTypes.DEVICE_TAGS_REQUEST, message);
       }
     });
     ```
3. **Update the UI** to handle and display only the returned subset.  

---

**Note**: Feel free to add labels such as `enhancement`, `performance`, or `help wanted`.

unocelli, 2024-12-24:
Hi, Thanks for the report

eduardoexact, 2024-12-24:
Another observation I made is that when I split the same number of tags in multiple MQTT connections, the system is much more responsive.

henjoe, 2024-12-27:
For reference, how many tags do you subscribe to in the broker?

eduardoexact, 2024-12-27:
I have subcriptions and publish tags. In total there are more than 8000. The server is runing on a Raspberry pi 4. I was able to continue working by splitting the tags in separate coneections.