frangoteam/FUXA #1550: [FEATURE]

shirk12138, 2024-12-24:
Hello, I am drawing a larger image with a size of approximately 16mb. When I want to import JSON files, the system prompts' View wrong format '. I have tried importing smaller JSON files without any problem. Have you configured the import size anywhere in the system? What should I do?

eduardoexact, 2024-12-24:
I faced problem parsing big Json Projects. In my case this solved the issue:

Edit the `FUXA/server/main.js` file and replace this:

```
const express = require('express');
const app = express();
```

With this:

```
const express = require('express');
const bodyParser = require('body-parser');

const app = express();

app.use(bodyParser.json({ limit: '100mb' }));
app.use(bodyParser.urlencoded({ limit: '100mb', extended: true }));
```

unocelli, 2024-12-24:
Hi, you can try to set the commented settings in `FUXA/server/_appdata/settings.js`

apiMaxLength: '15mb',

shirk12138, 2024-12-25:
No, it's ineffective, it's still the original problem

unocelli, 2025-01-10:
If by json file you mean the project, setting apiMaxLength in settings.js works as expected...I just tested it!...restart the server and if the problem persists maybe it's not in fuxa you need to look for.

unocelli, 2025-01-16:
Are you trying to import a View or open the project? They are two different jsons and you have to use two different fields.

unocelli, 2025-03-04:
I assume the issue has been resolved in the meanwhile. I'm therefore going to close this issue.