frangoteam/FUXA #1556: [FEATURE]

Wall-E1271, 2025-01-03:
**Describe the feature**
After you confirm the alarm, when the process value returns to the normal range again (there is no alarm value at this time), the value of "toack" will change to 1, and then immediately 0 again, which will generate an alarm. I want it don't change to 1 again, because it will disturb operator.

**Describe the solution you'd like**
Let the "toack" to stay 0.

**Additional context**
no

unocelli, 2025-01-04:
Hi, you mean that it would be a feature to be able to disable an alarm for a specified time.

Wall-E1271, 2025-01-10:
> Hi, you mean that it would be a feature to be able to disable an alarm for a specified time.

I'm sorry my description is not clear, what I mean is that I need to quote the value of toack of api/alarms to judge whether I need to generate an audible alarm, when the process value exceeds the alarm value, the value of toack will change from 0 to 1, this is an audible alarm, after confirming the alarm, the value of toack becomes 0, and the alarm will not continue to be generated. However, when the process value returns to the abnormal value, the value of toack will change to 1 for a short time, and then change back to 0, so that the alarm sound will be sounded again. What I want is that when the process value returns to normal, the toack stays at 0, and then changes to 1 when the next process value exceeds the alarm value again.

unocelli, 2025-01-15:
It seems to me that what you have described is the reaction of an alarm for which the whole flow (activ-passiv-ack) takes place at two different times so it is two alarms. i think you should try to handle this in the same script where you activate the audiable alarm, e.g. by managing a list of alarms already triggered in a certain interval.

Wall-E1271, 2025-01-18:
Thank you, I'll try to do it.