frangoteam/FUXA #1559: To restore

KWYHDHJ, 2025-01-08:
**Describe the bug**
Now I want to understand how the 'To restore' corresponding to the previous configuration tags is used and how to view the effect.
[image]

KWYHDHJ, 2025-01-09:
Ok, thank you very much

unocelli, 2025-01-09:
this feature is typically used for settings (Setpoints) to be saved in a PLC, when restarting the system.

TomasAdvance, 2025-02-19:
Hi, May I know, how to working if we have selected InfluxDB as DAQ. I investigation was found, if DAQ is base on Influxdb, restore function not working properly.
Only if SQLitle is selected, then restore behavior work properly.
May I ask you to verify it?
THX

unocelli, 2025-02-28:
Hi, Thanks for the report, I think I was able to fix a mistake

unocelli, 2025-03-04:
I’m going to close this as resolved. let me know if you have any issues.

closed 2025-03-04 by unocelli