frangoteam/FUXA #1561: [BUG]  Cannot read properties of undefined (reading 'numberOfItems')

sweetwisdom, 2025-01-09:
**Describe the bug**
A clear and concise description of what the bug is.

**To Reproduce**
```
ERROR TypeError: Cannot read properties of undefined (reading 'numberOfItems')
    at Object.convertPath (fuxa-editor.min.js:18:18787)
    at $.SvgCanvas.svgToString (fuxa-editor.min.js:18:22536)

```

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

[image]

## demo project

[fuxa-project (2).json](https://github.com/user-attachments/files/18359228/fuxa-project.2.json)

unocelli, 2025-01-10:
Try the last master branch version, should catch the error

unocelli, 2025-03-04:
I assume the issue has been resolved in the meanwhile. I'm therefore going to close this issue.

closed 2025-03-04 by unocelli