frangoteam/FUXA #1567: [BUG] Tag updation gets stops after Session keep Alive failure in OPCUA

Kiruthick-Roxx, 2025-01-13:
**Description**
When the OPC UA session encounters a "Keep Alive" failure, subsequent attempts to update tags fail. This issue persists even after the session is reconnected, causing disruptions in data synchronization.

**Steps to Reproduce**

[image]

- Establish an OPC UA session and start updating tags.
- Simulate a "Keep Alive" failure by disrupting the network or server connection (atleast more than a hour).
- Reconnect the session after the failure.
- Attempt to update tags.
- Expected Behavior:
- Tag updates should resume normally after the session is reconnected.

**Actual Behavior**
Tag updates fail even after the session is reconnected, requiring a complete restart of the server to restore functionality.

**Environment**
 - FUXA version: V.1.2.3-2105
 - Node.js version: 18.20
 - npm version: 10.9.0
 - Platform/OS: Ubuntu 20.04
 - Browser: Chrome

unocelli, 2025-01-28:
Hi, have you try to investigate how to solve the issue?

Kiruthick-Roxx, 2025-02-03:
Hi @unocelli ,
Once this error occurs, it does not come out this error and gets stuck in a loop, even though when the device is powered up again and communicate. The solution that worked for me it restart FUXA server or disable that particular device and enable it again.

I think a good solution would be to check for a number of session keepalive failure (say 10) and then disconnect the session and reconnect it, also create a toaster for the same as this also accounts for  device connection failure.

unocelli, 2025-02-14:
Great suggestion! We invite you to contribute directly with a Pull Request.

Kiruthick-Roxx, 2025-02-15:
Hi @unocelli,
I'm working on it.