frangoteam/FUXA #1568: [BUG] After initialization, click the Settings pop-up menu and then click View to report an error

MorganMaohong, 2025-01-13:
**Describe the bug**
When I create or open a new project, without doing any operations, I click Settings and click the View button, and an error message appears.

**To Reproduce**
Steps to reproduce the behavior:
1. Open the settings pop-up window
2. Click the View button
3. An exception occurs
4. ```main.d1fe082e6468ffe7.js:3 TypeError: Cannot read properties of undefined (reading 'setAttribute')
    at $.SvgCanvas.setBackground (fuxa-editor.min.js:20:1044)
    at r (fuxa-editor.min.js:21:6358)
    at u.init (fuxa-editor.min.js:22:4428)
    at r.myInit (main.d1fe082e6468ffe7.js:3:368757)
    at r.ngAfterViewInit (main.d1fe082e6468ffe7.js:3:367171)
    at qe (main.d1fe082e6468ffe7.js:329:2884705)
    at _t (main.d1fe082e6468ffe7.js:329:2884518)
    at Is (main.d1fe082e6468ffe7.js:329:2884231)
    at Ou (main.d1fe082e6468ffe7.js:329:2918078)
    at TE (main.d1fe082e6468ffe7.js:329:2917478)
    myInit @ main.d1fe082e6468ffe7.js:3Understand this errorAI
    main.d1fe082e6468ffe7.js:329 ERROR TypeError: this.winRef.nativeWindow.svgEditor.setColor is not a function
    at r.setFillColor (main.d1fe082e6468ffe7.js:3:376635)
    at r.myInit (main.d1fe082e6468ffe7.js:3:368814)
    at r.ngAfterViewInit (main.d1fe082e6468ffe7.js:3:367171)
    at qe (main.d1fe082e6468ffe7.js:329:2884705)
    at _t (main.d1fe082e6468ffe7.js:329:2884518)
    at Is (main.d1fe082e6468ffe7.js:329:2884231)
    at Ou (main.d1fe082e6468ffe7.js:329:2918078)
    at TE (main.d1fe082e6468ffe7.js:329:2917478)
    at Ou (main.d1fe082e6468ffe7.js:329:2917504)
    at IE (main.d1fe082e6468ffe7.js:329:2924710)```

**Screenshots**
[image]

**Environment**
 - FUXA version: 1.2.3-2125
 - Node.js version: v18.20.5
 - npm version:10.8.2
 - Platform/OS:Mac OS
 - Browser:chrome

unocelli, 2025-01-16:
Hi, I tested with the last version of master branch and I can't reproduce the issue.

unocelli, 2025-03-04:
I assume the issue has been resolved in the meanwhile. I'm therefore going to close this issue.

closed 2025-03-04 by unocelli