frangoteam/FUXA #1569: a problem

Hu2Zhao, 2025-01-14:
please help me!

[image]

**Environment**
 - FUXA version: 1.2.3
 - Node.js version: 18.18
 - npm version:
 - Platform/OS:
 - Browser:

Hu2Zhao, 2025-01-14:
Version 1.1.7 is right

unocelli, 2025-01-15:
Hi, try to convert the value with the Read value script in Tag Options

Hu2Zhao, 2025-01-17:
Thank you!! Can you give an example? I couldn't find how to set it up。

[image]

unocelli, 2025-01-28:
[wiki](https://github.com/frangoteam/FUXA/wiki/HowTo-Devices-and-Tags)

unocelli, 2025-03-04:
I assume the issue has been resolved in the meanwhile. I'm therefore going to close this issue.

closed 2025-03-04 by unocelli