frangoteam/FUXA #1572: [BUG] Cannot read properties of null (reading 'clientX')

CYO9000, 2025-01-15:
Hey first of all thank you so much for this awesome project! I really enjoy working with fuxa on my projects and never had major issues while doing so. However yesterday i encountered a strange issue when trying to open a card via $opencard in script. Today i tried resolving this issue by installing different versions of fuxa, trying different browsers and looking in code with the help of chatgpt. Unfortunately i wasnt successful and im stuck with this issue. Does anybody have an idea what causes this? Here is an example project which causes this issue on my end:
[test.json](https://github.com/user-attachments/files/18429084/test.json)

**Describe the bug**
When running a script which uses $opencard nothing happens and an error is reported in console.

**To Reproduce**
In the attached project click the button on MainView.

**Screenshots**
[image]

**Environment**
 - FUXA version: 1.2.3-2106
 - Node.js version: 18.20.5
 - npm version: 10.8.2
 - Platform/OS: Windows 10
 - Browser: Chrome/Edge/Firefox

unocelli, 2025-01-16:
Thanks for the report

CYO9000, 2025-01-18:
I installed the master-branch and it works as expected thanks a lot!

unocelli, 2025-01-19:
@mauroalexandre Sorry, I forgot to notify you that I am in charge of it. Thanks.

unocelli, 2025-03-04:
I’m going to close this as resolved. let me know if you have any issues.

closed 2025-03-04 by unocelli