frangoteam/FUXA #1577: sending commands via MQTT

MrKelser, 2025-01-17:
Hi, I have a question.
I need to turn on/off a relay via MQTT, I am using mosquitto broker.

mosquitto_pub -h localhost -p 1883 -t 'application/fe7bae52-4682-497c-a62c-7830637193e4/device/a8404152d185b432/command/down' -m '{"devEui":"a8404152d185b432","confirmed":true,"fPort":5,"data":"AwEB"}'

With this command, I can turn on the relay from the terminal, but subscribing from FUXA does nothing. I would like to activate it with a button, but I don't know how.

[https://i.postimg.cc/HsXCnsK6/imagen.png](url)

unocelli, 2025-01-28:
Hi, you have to publish and not subscribe, then bind the the Tag 'reles on' to a button event

unocelli, 2025-03-04:
I assume the issue has been resolved in the meanwhile. I'm therefore going to close this issue.

closed 2025-03-04 by unocelli