frangoteam/FUXA #1579: [BUG]

chaosyoh, 2025-01-19:
**Describe the bug**
switch button Set 'On Value' 1,  but value in device dont change in some case( is excepted ). the switch button always at On position, but the vlaue in device is 0 in fact, and it won't update

**Expected behavior**
switch button update UI in the next reading from device even if the value is same as the previous value

**Screenshots**
[image]

**Environment**
 - FUXA version: 1.2.3-2139
 - Node.js version: 20.04
 - npm version: 10.9.2
 - Platform/OS: windows
 - Browser: edge

unocelli, 2025-01-28:
Hi, I'm not sure if is a problem the nodejs version (supported is 18) and eventually we need more info

chaosyoh, 2025-02-05:
Think you for your reply. I try again in  the nodejs version 18, but sadly it happened again.  
My device is Siemens S7-1200, when I click the switch button, it send the value 1 to plc, but the tag in plc won't change in a scan cycle because of some reason, see more detail in the following pic
program in Siemens S7-1200:

[image]

In my Siemens HMI, it will switch to OFF position quickly after a click, I except the same in FUXA,

unocelli, 2025-02-14:
We can not reproduce the issue

unocelli, 2025-03-04:
I assume the issue has been resolved in the meanwhile. I'm therefore going to close this issue.

closed 2025-03-04 by unocelli