frangoteam/FUXA #1584: Leaflet or google maps implementation

MrKelser, 2025-01-22:
Is it possible to add maps like Leaflet or Google Maps into FUXA in any capacity?
I am using some devices that send their GPS location and it would be useful to put those coordinates into a map.

henjoe, 2025-01-27:
I have the same use cases that uses Maps. 
This will be a good features. 
@unocelli

unocelli, 2025-01-28:
Hi, it would be great to have this feature (#174). has anyone investigated which library can be used, google maps costs!

henjoe, 2025-01-29:
@unocelli  Leaflet is actually opensource and we can integrate/download it on Angular.

https://javascript.plainenglish.io/a-simplified-journey-of-leaflet-js-integration-with-angular-step-by-step-guide-b223f19b71f0

i am thinking this is similar to iframe where the leaflet map is wrapped on iframe, but have the options to add icons,marks and etc.

MatthewReed303, 2025-01-29:
for map integration it needs to be done correctly so you can interact with the map also, click on markers to then display site information or pop ups etc be able to draw geo fencing on to the map. I think leaflet is the best choice and open street maps, if user needs google maps overlay, maybe we have an option to select that and they enter in there API key for google?

henjoe, 2025-01-30:
Then the Marker can be binded to actions (OpenCard, setValue and etc.) when clicked. 

And also the Marker can be a type "Text" and be binded to a value of Tag.

henjoe, 2025-02-19:
Hi @unocelli  can you show how to use the maps? I wanna test it. 

1. How to set start location
2. How to add Marker
3. How to add popups

MatthewReed303, 2025-02-19:
@henjoe it's quite easy... add a new view and change view type to map. Then right click the map and add location and set your start location. This is just very basic right now. Hopefully in the future we can change layers, set default zoom levels, draw polygon groups on map, custom icons etc but it's a very good start and proof of concept.

henjoe, 2025-02-19:
> @henjoe it's quite easy... add a new view and change view type to map. Then right click the map and add location and set your start location. This is just very basic right now. Hopefully in the [...]

Thanks Matt! 

I will try this and commit to report if there were bugs and suggest some features. 

This is actually a great start. 
Kudos to the dev team!

henjoe, 2025-02-20:
Hi, tried and tested it. 
It works fine and good.

[image]

Just have one question, what is the "View" is for?

MatthewReed303, 2025-02-20:
@henjoe create a second view in Fuxa and set the view to that new view and then when you click the map marker it will open the view as a pop up

henjoe, 2025-02-20:
Yeah! This is cool!

[image]

Is it really transparent?

unocelli, 2025-02-21:
New you are able to set Goto Page and Open Link in a new Tab:

[image]

[image]

henjoe, 2025-02-22:
> New you are able to set Goto Page and Open Link in a new Tab:
> 
> [image]
> 
> [image]

This is really cool! ✅🥳

Hope we can pass options in the the View or Card for VariablesMapping ✌️

2veinte, 2025-03-03:
What version of FUXA are you using? On v1.2.3-2106 i can't change the view type to map

[image]

killerbeat12, 2025-06-26:
This is great! how could i get the latitude and longitude from signals and dynamically fill the map with this?

unocelli, 2025-07-09:
@killerbeat12 Currently, this feature is not yet supported
Feel free to open a new issue for it. I’d also love to hear more about your use case.