frangoteam/FUXA #1590: [QUESTION]: websocket usage for data updates

kheinrich188, 2025-01-28:
Hi there,

I’m experiencing an issue when trying to use the new WebSocket option in FUXA to update data more frequently than the 1-second interval provided by the WebAPI.

Here’s what I’ve done:

1. 	Switched my WebAPI connection to an “internal” connection to retain the tags.
2. 	Created a server-side script to fetch the initial data using the WebAPI.
3. 	Set up the WebSocket connection to handle subsequent updates.

However, the behavior isn’t as expected. When fetching tag data by ID using `$getTag()`, it always returns `null`.

Example API Response:

From my WebAPI:

```
{
    "t_8f60fd-f8073b": "1",
    "t_cb52db-724556": "3",
    "t_a2d094-5596cc": "1"
}
```

The WebSocket returns the following:

```
42/fuxa,
[
    "fuxa-data",
    {
        "tagId": "t_9bece6-11c1c5",
        "data": "-21.2713623046875"
    }
]
```

When I try to then `setTag(tagId, data)` I always get `setValue not supported!`

Despite successfully retaining tag IDs and verifying their values using the template function `$getTag()`, attempting to access tags via `$getTag('t_8f60fd-f8073b')` consistently returns `null`.

`console.log($getTag('t_8f60fd-f8073b' /* stHMIInterface - iCurrentUnit */));`

Environment:
FUXA Version: `1.2.3`

Expected Behavior:

- Only have an internal device with tags. 
- `$setTag` to set the data to a internal device tag to get rid of the webapi connection

Please let me know if I’m missing something in the setup or if this is a bug. Any guidance would be much appreciated!

Thank you!

kheinrich188, 2025-01-28:
@MatthewReed303 I saw it and follow the steps but for any reason I t will not work to set the tags. Did you set your tags as an internal connection?

MatthewReed303, 2025-01-28:
@kheinrich188 I used the Fuxa Server Tags and OPC-UA connection, any reason for internal? That's front end only, so only in the clients web browser.

kheinrich188, 2025-01-28:
Ok so I need to keep the WebAPI Connection as it is. Then I do a 1s get update of all data until the script and websocket connection is established then I think I can stop the device fetching every 1s and update the data with the websocket data?

kheinrich188, 2025-01-29:
As I see it is not allowed to set values when the connection has an address right? 

```
 this.setValue = async function (tagId, value) {
        if (apiProperty.ownFlag && data.tags[tagId]) {
      ....
       }
}
```

@unocelli can you help?

unocelli, 2025-02-14:
@kheinrich188 Hi, sorry but I got lost, I didn't understand what you are trying to do

kheinrich188, 2025-02-15:
@unocelli I want to Pull initial my data via WebAPI and then stop pulling the data every x seconds, instead I want to open a websocket connection to my server via script and update there only changing data fields. That would improve the performance and the data can be get faster updates < 1s

unocelli, 2025-03-04:
I'm not sure to understand your scope, FUXA send only current View data, If you will send only updated value then you have to manage and check the unchanged data in frontend in different case

unocelli, 2025-05-04:
Closing this due to inactivity. If the issue is still relevant or unresolved, feel free to reopen.

closed 2025-05-04 by unocelli