frangoteam/FUXA #1594: OPC UA server variable is not displayed in the editor output value field

Rainproof, 2025-01-30:
Hi.
1. I created an OPC UA server in FUXA
2. I created an output value field in editor
3. I assigned a tag from  OPC UA server to the field
4. I set the value in the my OPC UA server, but the value does not appear in the field. The value from OPC UA server appears in the Settings --> Connections --> OPC UA server --> tag
5. The same output value field displays the values ​​from the modbas tsp register

[image]
[image]
[image]
[image]

Rainproof, 2025-01-31:
I tested other data types via the OPC UA demo server. The described problem is observed only with int64. For some reason, the integer value int64 is displayed in Fuxa as a decimal fraction.

unocelli, 2025-01-31:
Hi, nodejs don't know 64bit type as number but as BigInt and currently is in FUXA not supported, the value will be converted to array of two 32bit, you can use the Read/Write Scala script to convert it. I will use only value[1]

Rainproof, 2025-01-31:
That is, the number int64 Fuxa represents in the form [second int32].[first int 32]?
For example, 88 is 0.88? Where 0 is the second int32, and 88 is the first int32?

closed 2025-03-04 by unocelli