frangoteam/FUXA #1596: datetime picker for the table element

Rainproof, 2025-02-02:
Hi,
Is there a function in Fuxa to select date and time for the table element?

unocelli, 2025-02-04:
Hi, you can configure the row event selection that open a dialog with a input control defined as datetime.

Rainproof, 2025-02-12:
Hi,
I'm looking at the code now:
D:\FUXA\client\src\app\gauges\controls\html-table\data-table\data-table.component.html
It contains the lines:
<div *ngIf="tableOptions.daterange.show" class="date-range">
Does this mean that the function can be selected by time and date?
But when I create a "table" element, I don't find such options for this element in the editor.

You write:"you can configure the row event selection that open a dialog with a input control defined as datetime."
But in these settings I only see the ability to run a script

[image]

unocelli, 2025-02-13:
Hi, the select is a event select of the user, that if a user select a row then start a script.
Select row from a script is missing.

unocelli, 2025-03-04:
I assume the issue has been resolved in the meanwhile. I'm therefore going to close this issue.

closed 2025-03-04 by unocelli