frangoteam/FUXA #1599: FUXA gets very slow and can access from browser

MrKelser, 2025-02-03:
**Describe the bug**
I have been working on Fuxa and I have reached a point (about 5 scripts and 6 mqtt devices) where the server does not refresh, save, or even enter from a browser. The server (laptop with ubuntu server 22.04LTS) where I run FUXA also runs mosquitto mqtt broker and chirpstark and these two programs run with no problem.

**To Reproduce**
Steps to reproduce the behavior:
1. get about 5 scripts, 6 mqtt clients, 15 views (6 for charts)
2. try to save, refresh, or even enter from a browser.
3. Long time response or not even working

**Expected behavior**
save, refresh, or accessing 

**Environment**
1. I made the installation with docker-compose.
2. Fuxa runs on Ubuntu 22.04LTS
3. I access it from Windows with Firefox.

unocelli, 2025-02-07:
Hi, can depend on several factors and needs more detail...check the logs. Disable broadcasting in Settings.
How many tags have you configured?

unocelli, 2025-10-03:
I assume the issue has been resolved in the meanwhile.