frangoteam/FUXA #1600: [BUG] Modbus Bitmask operation

trayuser, 2025-02-06:
**Describe the bug**
Using FUXA-1.2.3 on windows, node v22.13.1

The bit set works for the first bit eg bit 0,
But when I use the second bit eg bit 1 (or a higher bit index_.
Nothing is written to the modbus device (process controller)
The error log has a writeMemory error

2025-02-04T08:00:59.636Z [ERR] 'modbus' _writeMemory error! [object Object]

If I use the event tab to specify a bit (1) to toggle the controller value is updated but the toggle switch in FUXA doesnt update.

[fuxa-devices_mod_tags_20250206_01.csv](https://github.com/user-attachments/files/18686522/fuxa-devices_mod_tags_20250206_01.csv)

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected behavior**
The bit specified in 

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Environment**
Using FUXA-1.2.3 on windows, node v22.13.1

 - FUXA version: 
 - Node.js version: 
 - npm version:
 - Platform/OS:
 - Windows 10
 - Browser:
Firefox 128.4.0esr
[image]

[image]

unocelli, 2025-02-14:
The Bitmask is a mask of the value, in your switch you have configured to 1, for bit 2 you have to configure to 2.

Please provide a project to reproduce the issue

trayuser, 2025-02-15:
Hi,

Please see attached

[fuxa-project_V01_20250216_01.json](https://github.com/user-attachments/files/18812959/fuxa-project_V01_20250216_01.json)

unocelli, 2025-03-04:
Hi, change the modbus address and try it
[fuxa-project - 2025-03-04T205654.515.json](https://github.com/user-attachments/files/19076895/fuxa-project.-.2025-03-04T205654.515.json)

trayuser, 2025-03-06:
Yup using the supplied project file with IP address adjusted to local IP address of my controller. 
The project works with bit mask feature.  

Looks like operator error  at my end  ;)
Thank you.

PS: To correct the operation, load the bit value in the ON Value in the dialog box eg bit 7 is 128

[image]