frangoteam/FUXA #1602: [BUG] Unable to select pipe after adding SVG in and position gets changed automatically in a view

Kiruthick-Roxx, 2025-02-08:
**Describe the bug**
After adding an SVG file to pipe and setting animation speed, the SVG does not get loaded on the first try. After refresh, it works and even after adding the SVG, the pipe is not getting selected for moving. Sometimes, even the position gets changed automatically and dragging the pipe using cursor also does not work properly.

It would be good that the image should only be moving, and at start of the pipe, a static image is visible. 

**To Reproduce**
Steps to reproduce the behavior:
1. Choose pipe control and draw a pipe
2. Click on add SVG image and add new SVG File
3. Now check if the SVG is added.
4. If added, try to select the file.
5. Now you can see that, you cannot select and move the pipe.

**Expected behavior**
It should load the SVG in the first try itself and also should be draggable and clickable

**Screenshots**

https://github.com/user-attachments/assets/101c3740-d12e-49b8-b850-dab61b752b93

https://github.com/user-attachments/assets/ede74fe2-284c-4efa-99d1-9fd87328f3ee

**Environment**
 - FUXA version: FUXA V.1.2.4-2161
 - Node.js version: 18.20.4
 - npm version: 10.9.0
 - Platform/OS: Ubuntu 20.04
 - Browser: Chrome version: 132.0.6834.160

unocelli, 2025-02-14:
Hi, please provide your project

Kiruthick-Roxx, 2025-02-15:
Hi @unocelli,
I have attached my project here. It is possible to remove the initial display of the SVG as you can see in the project, so that it will be realistic to animate. Like appear, move and disappear.

[fuxa-project (2).json](https://github.com/user-attachments/files/18809631/fuxa-project.2.json)

unocelli, 2025-02-18:
please, can you send me the svg image alone?

Kiruthick-Roxx, 2025-02-19:
Hi @unocelli,
      I have attached the SVG here.

[image]

unocelli, 2025-02-27:
Hi, the feature is designed for simple svg (without scripts), we will see about implementing a filter to prevent such issues

Kiruthick-Roxx, 2025-03-06:
Hi @unocelli,

Thanks. It works with simple SVG.

closed 2025-03-25 by Kiruthick-Roxx