frangoteam/FUXA #1604: [BUG] Loss of Tag Connections

robsori, 2025-02-12:
Hi. While working with Fuxa, I encountered a rather frustrating issue: sometimes, changes made to a project are not saved. Additionally, there are instances where certain objects lose their connections to tags. I haven't yet identified the exact conditions under which this happens.
A simple refresh of the web page used for configuration is not enough to fix the issue. The only solution I've found so far is to close and reopen the browser.
Has anyone else experienced this problem? If so, have you found a solution?

**Environment**
 - FUXA version v1.2.3: 
 - Node.js version 18.20.4: 
 - npm version: 6.14.18
 - Platform/OS: Windows 10 Pro
 - Browser: Chrome

unocelli, 2025-02-13:
Hi, sounds a bit vague to me, do you have a project to share?

robsori, 2025-02-15:
Hi. I don't have a specific project available. It's not an error I can reproduce, it happens sometimes while working. I see that someone else has encountered this problem: https://github.com/frangoteam/FUXA/discussions/1465

unocelli, 2025-10-03:
Check Allocated memory of browser with DevTools