frangoteam/FUXA #1608: [BUG] ModbusTCP no values with adress ofset 42999 and higher

egj162, 2025-02-16:
**Describtion**
ModbusTCP connetion with adress ofset 42999 and higher do not get a value. 
We have a ModbusTCP device with adress ofset > 42999 but for these we do not get values with FUXA. We already tested the same adress with tools like "Modbus Client X".  

**To Reproduce**
Steps to reproduce the behavior:
1. Make a new connection via ModbusTCP
2. Add a value with adress ofset like 42999
3. You See no value 

**Screenshots**

[image]

[image]

[image]

**Environment**
 - FUXA version: 1.2.3-2106
 - Node.js version: v18.13.0
 - npm version: 9.2.0
 - Platform/OS: Ubuntu 23.10
 - Browser: Mozilla Firefox 135.0

martinjug, 2025-02-17:
Hi,  

The standard Modbus register ranges are as follows:  

- **0x** = Coil → **00001–09999**  
- **1x** = Discrete Input → **10001–19999**  
- **3x** = Input Register → **30001–39999**  
- **4x** = Holding Register → **40001–49999**  

When selecting the **Holding Register (4x)** type, the system automatically assumes the **40000** range. Therefore, you only need to specify the offset from **40001**:  

- **43000 → 3000**  
- **43001 → 3001**

egj162, 2025-03-06:
Sorry for that late answer.
@martinjug 
In your picture you can see there ist mentioned 400 000 and nocht 40 000 for the holding register. So the offset 43 000 has to be specified complete. In my example other values are working correct liekt the first one in my picture (offset 41 194).