frangoteam/FUXA #1612: [BUG] Cannot read properties of null (reading 'toString') - Reports Generation

eyadalhanafi, 2025-02-20:
Describe the bug:
When we tried to generate the report, it failed, and the following error occurred: "Cannot read properties of null (reading 'toString')." The system is currently in production, and this issue surfaced after we upgraded to the latest version.

 - FUXA version:  1.2.4-2178
 - Node.js version: v18.9.1.
 - Platform/OS: Windows 10.0
 - Browser: Google Chrome

[image]

eyadalhanafi, 2025-02-20:
after investigation, it seems the error is possibly due to **imageGenerator.createImage** returning **null**, which causes the **toString('base64')** call to fail.

unocelli, 2025-02-21:
Hi, you have to install chart.js@2.9.4 and chartjs-node-canvas@3.2.0...I just tested and works as expected

eyadalhanafi, 2025-02-22:
We have the following server plugins installed

[image]

eyadalhanafi, 2025-02-25:
The issue has been resolved. I appreciate your advice!

unocelli, 2025-03-04:
I’m going to close this as resolved. let me know if you have any issues.

closed 2025-03-04 by unocelli