frangoteam/FUXA #1613: Help in setting Serial Port in Ubuntu Linux

GopalN, 2025-02-20:
I have installed the latest version fuxa and trying to communicate using Modbus RTU.I am  using a USB to TTL converter and is visible on /dev/ttyUSB0 .I am using a docker image and have run it using sudo still I am getting this in the log

2025-02-20T15:28:43.237Z [info] 'Modbus' try to connect /dev/ttyUSB0
2025-02-20T15:28:43.240Z [info] daqstorage.bind-daq-data '_db/daq-data_d_bced84f9-cc944e9d_20250220102843.db' database
2025-02-20T15:28:43.240Z [info] daqstorage.bind-daq-map '_db/daq-map_d_bced84f9-cc944e9d.db' database
2025-02-20T15:28:43.241Z [error] 'Modbus' connect failed! Error: Error: No such file or directory, cannot open /dev/ttyUSB0
2025-02-20T15:28:43.242Z [error] 'Modbus' connect error! undefined (0)

Can  anybody help me in solving this

MatthewReed303, 2025-02-20:
@GopalN  you have to pass through the serial port/USB via docker. I use Docker Compose for this, for example:

```
version: '3.3'

services:
    fuxa:
        image: frangoteam/fuxa:latest
        network_mode: "host"
        #ports:
        #    - 1881:1881
        devices:
            - "/dev/ttyS0:/dev/ttyS0"
        volumes:
            - ./fuxa_appdata:/usr/src/app/FUXA/server/_appdata
            - ./fuxa_db:/usr/src/app/FUXA/server/_db
            - ./fuxa_logs:/usr/src/app/FUXA/server/_logs
            #- ./fuxa_images:/usr/src/app/FUXA/server/_images
        environment:
            - TZ=Etc/UTC
        restart: always
```

In Fuxa device make sure to set the as in docker file `/dev/ttyS0` looks like in your case you need to pass through `/dev/ttyUSB0`

GopalN, 2025-02-21:
This has solved my problem .Thank you @MatthewReed303

closed 2025-02-21 by GopalN