frangoteam/FUXA #1617: [BUG]script error:document is not defined

pystudy123, 2025-02-22:
**Describe the bug**
I want to click a button in fuxa to pop up an iframe pop-up window, but it has no effect.

**console**
ReferenceError: document is not defined
    at Object.script_2 (C:\fuxa2024\server\runtime\scripts\msm-scripts.js:78:83)
    at MyScriptsModule.runScript (C:\fuxa2024\server\runtime\scripts\msm.js:63:51)
    at C:\fuxa2024\server\runtime\scripts\index.js:77:43
    at new Promise (<anonymous>)
    at ScriptsManager.runScript (C:\fuxa2024\server\runtime\scripts\index.js:70:16)
    at C:\fuxa2024\server\api\scripts\index.js:41:36
    at Layer.handle [as handle_request] (C:\fuxa2024\server\node_modules\express\lib\router\layer.js:95:5)
    at next (C:\fuxa2024\server\node_modules\express\lib\router\route.js:144:13)
    at C:\fuxa2024\server\api\jwt-helper.js:66:17
    at C:\fuxa2024\server\node_modules\jsonwebtoken\verify.js:261:12
    at getSecret (C:\fuxa2024\server\node_modules\jsonwebtoken\verify.js:97:14)
    at module.exports [as verify] (C:\fuxa2024\server\node_modules\jsonwebtoken\verify.js:101:10)
    at verifyToken (C:\fuxa2024\server\api\jwt-helper.js:48:13)
    at Layer.handle [as handle_request] (C:\fuxa2024\server\node_modules\express\lib\router\layer.js:95:5)
    at next (C:\fuxa2024\server\node_modules\express\lib\router\route.js:144:13)
    at Route.dispatch (C:\fuxa2024\server\node_modules\express\lib\router\route.js:114:3)

**Environment**
 - FUXA version: fuxa-server@1.1.18-1460(I tested the latest version and had the same problem)
 - Node.js version:Node.js v18.20.2.
 - Platform/OS: windows server 2016
 - Browser: chrome

pystudy123, 2025-02-22:
script code:
`
    var iframe = document.createElement('iframe');
 
    iframe.src = 'https://cn.bing.com';  
    iframe.width = '600px'; 
    iframe.height = '400px';
    iframe.frameBorder = '0'; 
    
    var container = document.createElement('div');
    container.style.position = 'fixed';
    container.style.top = '0';
    container.style.left = '0';
    container.style.width = '100%';
    container.style.height = '100%';
    container.style.backgroundColor = 'rgba(0, 0, 0, 0.5)';
    container.style.display = 'flex';
    container.style.justifyContent = 'center';
    container.style.alignItems = 'center';
    container.style.zIndex = '9999';
  
    container.appendChild(iframe);

    document.body.appendChild(container);

    container.onclick = function() {
        document.body.removeChild(container);
    };
`

unocelli, 2025-02-22:
Hi, your script run in backend (server) document is in frontend!

unocelli, 2025-03-04:
I’m going to close this as resolved. let me know if you have any issues.

closed 2025-03-04 by unocelli