frangoteam/FUXA #1621: [QUESTION] "Connection Status" codes, what do they mean?

jeddors, 2025-02-24:
When you link a connection (e.g. MQTT server) Fuxa automatically generates a tag which represents the connection status:

[image]

I've played with stopping, starting, disconnecting the MQTT server and the tag value can be 0, 3, 5 etc.

I looked through the doc's but I can't find what any of these values actually mean

henjoe, 2025-02-25:
0 - means it is disconnected
5- means it is connected
3 - I am not sure but maybe idling status.

unocelli, 2025-02-25:
Hi,
var ConnectionStatusEnum = {
    OFF: 0,
    WARNING: 3, // up to 5 times the polling interval without response
    ON: 5,
}

closed 2025-03-04 by unocelli