frangoteam/FUXA #1626: [FEATURE] Filtering Incorrect Values  – Should Hysteresis Be Added?

Bundyra, 2025-02-26:
Hello

I have an issue with voltage readings in my SCADA system, which is connected to circuit breakers in an electrical switchboard. Occasionally, due to interference, the system registers incorrect values, reaching the maximum int16 value (32767). This results in false alarms and incorrect data in reports.

I am aware that a deadband function has been added for values, but in this case, it does not solve the issue. Deadband is useful for ignoring small fluctuations, but it does not protect against sudden spikes to the maximum value.

I am looking for an effective filtering mechanism that would ignore such erroneous values while still allowing real voltage changes to be recorded without delays.

Have you considered adding a hysteresis function for analog values in the SCADA system?
Is it possible to set acceptable thresholds so that voltage spikes are ignored only if they last for a short time?

I was considering using logic like:

- If a voltage reading suddenly jumps to 32767V, it should not be recorded immediately.
- The system should check whether the value persists for a few subsequent readings (e.g., 2-3 cycles).
- If the next readings return to normal (e.g., 230V), it should be considered noise, and the system should ignore the spike.

Bundyra, 2025-02-26:
[image]
I added this script to the fuxa, but ReadScaleScript remains blank. Any suggestion? My device is connected via ModbusRTU.

henjoe, 2025-02-26:
You have to create a script first with parameter value.

Bundyra, 2025-02-26:
Oh you are right, i changed the parameter name from "val" to "value" and it works. Thank you.

MatthewReed303, 2025-02-26:
@Bundyra  you could also try a simple low pass filter in scale section such as:

```
function lowPassFilter(newValue, prevFiltered, alpha) {
    return alpha * newValue + (1 - alpha) * prevFiltered;
}

// Example usage:
let filteredValue = 0; // Initial value
const alpha = 0.1; // Smoothing factor (0.0 - 1.0), lower = more filtering

filteredValue = lowPassFilter(newValue, filteredValue, alpha);
return filteredValue;
```

henjoe, 2025-02-27:
> @Bundyra  you could also try a simple low pass filter in scale section such as:
> 
> ```
> function lowPassFilter(newValue, prevFiltered, alpha, minValue, maxValue) {
>     newValue = Math.max(minValue, Math.min(newValue, maxValue)); // Clamp within range
>     return alpha * newValue + (1 - alpha) * prevFiltered;
> }
> 
> // Example usage:
> let filteredValue = 0;
> const alpha = 0.1;      // Smoothing factor (0.0 - 1.0), lower = more filtering
> const minValue = 0;
> const maxValue = 500; 
> 
> filteredValue = lowPassFilter(newValue, filteredValue, alpha, minValue, maxValue);
> return filteredValue;
> ```
> 
> You will have to adjust to suit your script, but will ignore large spikes and filter your readings

Never knew creating function inside script is possible. This is nice!

closed 2025-03-04 by unocelli