frangoteam/FUXA #1628: [BUG]incorrect display of nodes ops ua server

Rainproof, 2025-02-27:
**Environment**
 - FUXA version: 1.2.3
 - Node.js version: v18.19.1
 - npm version: v18.19.1
 - Platform/OS: Windows 10
 - Browser: Google Chrome

When I click on the node "cepn1", then the node "cepn"104 opens.
When I click on the node "cepn104", then the node "cepn104" opens again.
When I click on the node "lent1", then the node "lent116" opens again.
When I click on the node "lent116", then the node "lent116" opens again.

in UaExpert the nodes open correctly.

Details in the attached gif files.

UaExpert
[image]

FUXA 1.2..3
[image]

rusanz, 2025-02-27:
Same error on webApi, the method of addTreeNodes dont works. In my case i have modified this method to work with deep nested json, because for reason "x = potatto " if your node have parent and parent.parent and that parent.parent is Array  the node type is Item because yes. So if that node is a primitive  oh sorry you can´t select that node.

[image]

Rainproof, 2025-03-03:
I couldn't do anything better so far:
I corrected the line
**return result.sort((a, b) => (a.path > b.path) ? 1 : -1);**
to
**return result.sort((a, b) => a.id.localeCompare(b.id));**
in the file
_**client\src\app\gui-helpers\treetable\treetable.component.ts**_

I got this result:

[image]

Now the subnodes open correctly in their nodes.
But the subnodes of Objects open not under Objects.
I don't know how to fix this yet.

unocelli, 2025-03-09:
Hi, Thanks for the report.
is it possible to reproduce it with the UaCPPServer (my test bench)?

closed 2026-11-07 by unocelli