frangoteam/FUXA #1629: [BUG]

HMH1234567890, 2025-02-28:
**Describe the bug**
I successfully imported the SQLite3 library into the plugin, but when I actually ran the script, it prompted that the corresponding module could not be found（By the way, the FUXA I'm using is the basic version. Do I need to upgrade to the advanced version）

**To Reproduce**

[image]

[image]

unocelli, 2025-02-28:
Hi, you don't need to install a new, is already present in node_modules
`var sqlite3 = require('./node_modules/sqlite3').verbose();`

HMH1234567890, 2025-03-01:
[image]
At first, when I saw the console output, I thought it was because the module had not been loaded. 
           However, when I directly called this module, the console output was also the same.( The database file path is also fine)I don't quite understand what error the red part of the output in the figure is reporting. 
           When I used this code in the source code and directly placed it below the imported module for debugging, the IDE console output was normal

closed 2025-03-04 by unocelli