frangoteam/FUXA #1632: [BUG] Disappearing line chart.

Bundyra, 2025-03-03:
**Describe the bug**  
Graphs randomly disappear when the min and max values for the Y-axis are not defined. When selecting a specific time range, the values reappear.  

**To Reproduce**  
Steps to reproduce the behavior:  
1. Open the application and navigate to the dashboard.  
2. Add a graph without setting the min and max values for the Y-axis.  
3. Observe the graph over time—sometimes, it disappears (screenshot number 1).   
4. Select specific section of the chart (screenshot number 2).  
5. The graph reappears (screenshot number 3).  

**Expected behavior**  
The graph should always be visible, even when min and max Y-axis values are not explicitly set.  

**Screenshots**  
Screenshot 1.
[image]
Screenshot 2.
[image]
Screenshot 3.
[image]

**Example project**  
Please provide an example project where the issue can be replicated.  
- **Upload a `.zip` file** containing the minimal setup needed to reproduce the issue.  
- **Or attach the JSON file** of your project.  

**Environment**  
- **FUXA version:** v1.2.0
- **Node.js version:** 18.17.1
- **npm version:** 9.6.7 
- **Platform/OS:** Ubuntu 22.0.4.4 LTS  
- **Browser:** Google Chrome/Opera

unocelli, 2025-03-09:
Hi, we can't reproduce it, try the last version (1.2.4)

unocelli, 2025-05-04:
Closing due to inactivity.

closed 2025-05-04 by unocelli