frangoteam/FUXA #1635: How to display the results of adding, subtracting, multiplying, and dividing two variables in data display

alis963456, 2025-03-03:
How to display the results of adding, subtracting, multiplying, and dividing two variables in data display

unocelli, 2025-03-09:
Hi, you can use the script:
You can define a script with a 1-second interval that saves operations to a FUXA server tag. Similarly, you can define Read Scale scripts on variables for the calculation.

alis963456, 2025-03-10:
[image]Is this simpler and more convenient? We can perform direct calculations here, as well as simple operations such as addition, subtraction, multiplication, and division between two or more variables. When there is a large amount of calculated data on the page, it is obvious that the efficiency of the script is too low

unocelli, 2025-03-10:
I see, do you have a design of how to make an intuitive user interface?

alis963456, 2025-03-11:
[image]

[image]I think this design method is quite good. Clicking the button in the upper left corner allows you to select multiple points in the database, and then on the right side of the interface, you can do some simple calculations such as addition, subtraction, multiplication, division, AND or NOT, or compare sizes. You can refer to it

jingshui127, 2025-03-21:
感觉作用不大。
1.这样的需求不会很多，用脚本也不复杂。
2.也可以在数采网关实现。