frangoteam/FUXA #1636: [BUG]

Bundyra, 2025-03-04:
**Bug: Unable to Link Tags to Objects After Importing Devices**  

 **Bug Description**  
After importing devices using a template, tags from imported device appear in the object properties but cannot be permanently assigned. When a tag is added to an object and changes are saved, the tag disappears upon reopening the object properties. This prevents proper data linkage.  

**Steps to Reproduce**  
1. Import devices using a template.  
2. Open the object properties.  
3. Add a tag to the object.  
4. Save changes and close the properties window.  
5. Reopen the object properties – the tag is no longer assigned.  
 **Expected Behavior**  
Tags should remain assigned to the object after being added and saved.  

**Environment**  
- **FUXA version:** v1.2.3
- **Node.js version:** 18.19
- **npm version:** 9.2.0
- **Platform/OS:** Linux raspberrypi 6.6.74+rpt-rpi-v8 #1 SMP PREEMPT Debian 1:6.6.74-1+rpt1 (2025-                                                                                                                                                             01-27) aarch64
- **Browser:** Opera

unocelli, 2025-03-09:
Thanks for the report.
Hi @rikugun please can you check the issue?

rikugun, 2025-03-10:
> **Bug: Unable to Link Tags to Objects After Importing Devices**
>
> **Bug Description** After importing devices using a template, tags from imported device appear in the object properties but cannot be [...]

can you post your template for me ?

Bundyra, 2025-03-12:
Here you go. 

[template.json](https://github.com/user-attachments/files/19203261/template.json)

rikugun, 2025-03-13:
> Here you go.
> 
> [template.json](https://github.com/user-attachments/files/19203261/template.json)

hi @Bundyra  , i have try your file . but not reproduce the bug.

[image]

you can check your post data with devtool.

rikugun, 2025-03-13:
I see what you mean, I will fix the problem

commit 681c85f1 (2025-03-14): fix: #1636, change tags key to new id when use import device template
commit f6206584 (2025-03-14): Merge pull request #1650 from rikugun/bug/1636

fix: #1636, change tags key to new id when use import device template
closed 2025-03-14 by unocelli