frangoteam/FUXA #1642: [BUG]Performance problem while working with many snap7 devices

santa-han, 2025-03-07:
**Describe the bug**
We've been testing FUXA, it is amazing. You guys have done a great job.
Recently, we are planning to migrate one of our projects to FUXA.
There are more than 100 Siemens PLCs, each having around 200 tags.
When we import those devices into FUXA, lots of error/warning messages pop out, saying something like 'polling overload', and the tag values got updated very slowly (if they did).
I put logs in FUXA and node-snap7. It turns out, ‘s7client.DBRead’ was called first, and **after about 6 sec**, the real action 'opReadArea' inside node-snap7 was called.
I think this is a problem, but I'm not sure yet exactly what cause the delay.

**To Reproduce**
Steps to reproduce the behavior:
1. You have to have many real devices

**Expected behavior**
All tags for all devices update in time.

**Screenshots**
[image]

**Example project**
Please provide an example project where the issue can be replicated.
- **Upload a `.zip` file** containing the minimal setup needed to reproduce the issue.
- **Or attach the JSON file** of your project

[fuxa.log](https://github.com/user-attachments/files/19123061/fuxa.log)

**Environment**
 - FUXA version: 1.2.4
 - Node.js version: 22.14.0
 - npm version: 10.9.2
 - Platform/OS: 10/10
 - Browser: MS Edge

santa-han, 2025-03-07:
Also we've translate lots of texts into Chinese. hope i can commit them somtime.

unocelli, 2025-03-09:
Hi, 
Regarding the issue with polling overload and slow updates, the delay you're experiencing could be influenced by multiple factors, including system resource limitations and the way Siemens PLCs handle concurrent communication requests.

To better understand the root cause, we recommend performing some tests to determine at what point the issue begins to appear:
- Try adding PLCs and tags incrementally to identify when the delays start. This can help pinpoint whether the bottleneck is on FUXA, Node-Snap7, or the PLC side.
- Check CPU, memory, and network usage on the machine running FUXA. If the system is under high load, it may slow down request handling.
- Siemens PLCs have internal limitations on the number of simultaneous connections and data requests they can handle. If too many requests are sent at once, the PLC may introduce delays or drop some of them.
- Try increasing the polling interval or grouping tags into larger read operations to reduce the number of individual requests.
- Distribute the Load Across Multiple FUXA Instances, If you have a large number of PLCs, consider installing multiple FUXA instances on different machines and grouping PLCs across them. This can help balance the communication load and improve performance.

About the translation, please make a PR

santa-han, 2025-03-10:
> Hi, Regarding the issue with polling overload and slow updates, the delay you're experiencing could be influenced by multiple factors, including system resource limitations and the way Siemens PLCs [...]

Our old system is running on the same machine, and connects to the same PLCs. It woks fine. So it doesn't seem to be about the PLCs or the system resource. (*Of coz when testing FUXA, the old system was shut down.*)

> * Distribute the Load Across Multiple FUXA Instances, If you have a large number of PLCs, consider installing multiple FUXA instances on different machines and grouping PLCs across them. This can help balance the communication load and improve performance.

I thought about this. But our situation is, the customers want to see all the devices in one system. They wouldn't like to login to 10 systems. I don't know how to achieve the effect with multiple instances.

> 
> About the translation, please make a PR

Today I set up a OPC server with the same 100*200 tags configured. FUXA connects to this OPC server to read the tags. It was no good either. The tags didn't get updated in several minutes, and the browser often complained about "no response". (*I can understand the browser issue, coz I was browsing all the tags*.)

jingshui127, 2025-05-08:
我记得1.1.X的版本的时候，S7连接的变量好像不能草果30个，要不就经常掉线。所以后来我都是把现场所有PLC(西门子/三菱/OMRON等)都转成MQTT，FUXA只用MQTT。然后就比较稳定了。
最近测试1.2.4和1.2.5发现在windows系统下，S7还是非常不稳定，有时候一点反应都没有

unocelli, 2025-05-14:
Hi, please try using Node.js version 18, as it's the recommended version for better compatibility.

unocelli, 2025-10-03:
I assume the issue has been resolved in the meanwhile.