frangoteam/FUXA #1644: Advice on alarm Settings

andy8601, 2025-03-10:
Whether it is possible to add the alarm related Settings to the connected variable, set the alarm function on or off of this variable in the variable editing, and also set the information that needs to be alerted here; If the alarm information is entered later, the workload is huge.

henjoe, 2025-03-10:
That requires a lot of refactoring for the Tag variable. 
But that would be a good idea. 

Alarming options will be under the Tag Options -> other Tab let say "Alarming" and we can simply toggle on/off and set the tags setting on that specific tag.

henjoe, 2025-03-10:
One other way to do is when the Alarm is enabled on the Tag Options, it will automatically create an Alarm item with pre configured settings. What do you think @unocelli ?

unocelli, 2025-03-10:
@henjoe this implemantation need many change too. for both suggestions I am afraid that the ui becomes too chaotic

What you think about: allow that an alarm can be created without a binding to the tag and the trigger can be managed via script

andy8601, 2025-03-11:
My current project has about 40 groups, each group has about 60 variables that need to be alerted, and the variables that need to be alerted in each group are the same. There are also some alarms whose values range from 0 to 50, representing different alarm codes. The workload of setting these alarms separately is huge. How can I quickly deploy alarms? For example, after I set the alarm variable of a group, how to copy it? I've tried editing the database directly, but it still doesn't work.

MatthewReed303, 2025-03-11:
@unocelli what if we have an alarm import/export this would allow you to use an external python script etc to generate all your alarms in a CSV etc User could export there Tags as an CSV and use the script to generate those tags into alarms or manually do them in a CSV which would be a lot faster than using the current UI? 

Or add an simple CSV type editor in UI to create them quickly.

henjoe, 2025-03-11:
> @henjoe this implemantation need many change too. for both suggestions I am afraid that the ui becomes too chaotic
> 
> What you think about: allow that an alarm can be created without a binding to the tag and the trigger can be managed via script

It should automatically bind the Tag variable. Since it is created under Tag Options, we can get the current TagID, and simply put that ID in the Alarm setting tag variable. 

And from the Tag Options there is a button where they can click and it will show the Alarm settings UI. 
I think its doable and doesnt require too much refactoring.

andy8601, 2025-03-27:
In fact, the simplest interface UI for setting alarms is directly migrated to the page for setting variables, and when the "alarm" function is opened, the page UI for setting alarms appears. @unocelli @henjoe