frangoteam/FUXA #1653: [BUG]

Conger0392, 2025-03-18:
**Describe the bug**
When using widgets, the graphic's background is filled in white

**To Reproduce**
Steps to reproduce the behavior:

Add widget to canvas

**Expected behavior**
When using widgets, the background should be transparent just like normal SVG

**Screenshots**
[image]

**Example project**
Please provide an example project where the issue can be replicated.
- **Upload a `.zip` file** containing the minimal setup needed to reproduce the issue.
- **Or attach the JSON file** of your project

**Environment**
- FUXA version:1.2.5-2208(server/package.json)
- Node.js version:v18.20.5
- npm version:10.8.2
- Platform/OS:Windows 11 23H2
- Browser:chrome Version 134.0.6998.89 (Official version) (64-bit)

Conger0392, 2025-03-18:
[image]

MatthewReed303, 2025-03-18:
@Conger0392  can you try to set the background to none on the bottom left on the colour selector and then import the SVG/Widget

Conger0392, 2025-03-19:
Thank you very much to my friends, this is very helpful to me

[image]

unocelli, 2025-05-04:
I’m closing this issue since there is nothing feasible and due to inactivity.

closed 2025-05-04 by unocelli