frangoteam/FUXA #1655: [BUG]Questions about tag values and fill

Conger0392, 2025-03-19:
**Describe the bug**
One is that the label values do not match and the colors do not change.
Usually, I can fill the graph according to the value of the label, but if the label returns a mismatch, the color of the graph will remain as it was before.
For example: 0 is red, 1 is green, the graph is now green, and when the label returns 3, the graph is still green.
The other is to flag a return NaN match exception.
When the tag returns NaN, the fill judgment recognizes NaN as 0.I tried the bool and real Settings, and they both recognize NaN as 0.

**To Reproduce**
1.Set the graph to a range of 0-4 by filling red and green strokes, by dropping down the TAB or switching simulated values

[image]
2.Add a tag that returns NaN to test the match

[image]

**Expected behavior**
When label values do not match, fill in the default color or do not fill in

**Screenshots**

[image]

[image]

**Example project**
none

**Environment**
FUXA version:1.2.5-2208(server/package.json)
Node.js version:v18.20.5
npm version:10.8.2
Platform/OS:Windows 11 23H2
Browser:chrome Version 134.0.6998.89 (Official version) (64-bit)

unocelli, 2025-03-25:
Hi, try to insert as **first** condition: min -9999, max: 9999 color yellow

Conger0392, 2025-04-15:
> Hi, try to insert as **first** condition: min -9999, max: 9999 color yellow

Thanks, looks like it works

[image]

[image]

closed 2025-05-04 by unocelli