frangoteam/FUXA #1657: [FEATURE] Modbus holding register get bit by position in register

sergeyyarkov, 2025-03-20:
Hello, thank you for your awesome app. I'd like to propose a feature. In holding register I store bit values. For example, in the holding register of my buttons at address 400003 is stored the value 0b00000011_00000000 that says that two buttons are pressed. But when adding a tag to FUXA when I select the Boolean type I cannot select the bit position. As a solution I created script in fuxa: 

```
function bitPosition(value, position) {
return Number(!!(value & (0x01 << position)));
}
```

And it works but with UInt16 register type in FUXA

sergeyyarkov, 2025-03-20:
I also can't set the bit by position when the button is pressed

MatthewReed303, 2025-03-20:
@sergeyyarkov I'm pretty sure with Modbus you can't use bits in holding registers for bit level access because the bit's will get over written when the complete bytes are written to the register. You can use coils instead which is intended for this use?

sergeyyarkov, 2025-03-20:
> @sergeyyarkov I'm pretty sure with Modbus you can't use bits in holding registers for bit level access because the bit's will get over written when the complete bytes are written to the register. You can use coils instead which is intended for this use?  

Hi thank you for your answer. In TIA portal I am using the MB_SERVER function. So the Modbus coils will be physical outputs. And I'm not sure that writing to the output ports using a SCADA button would be the right solution from a spec standpoint

rvbatista, 2025-03-20:
> @sergeyyarkov I'm pretty sure with Modbus you can't use bits in holding registers for bit level access because the bit's will get over written when the complete bytes are written to the register. You can use coils instead which is intended for this use?  

In others scada write operations to bits of holding registers is possible, and the driver overwrite others bit with the same values from the last read.

MatthewReed303, 2025-03-20:
@rvbatista sorry yes you are correct, there is function 22 for mask write. I was not aware of this and Fuxa does does not support this. You could write your own script to read the bit register first and then send your bit update. 

@sergeyyarkov I have use Tia with Fuxa and ended upsetting up an OPC-UA Server on the PLC, the s7-1200 and s7-1500 both support this.

unocelli, 2025-04-20:
@sergeyyarkov if I understand you correctly, you would like to be able to put a bitmask when setting a click event with setValue to the button?

sergeyyarkov, 2025-04-20:
@unocelli hello. yes, exactly. and also if i can get the value by bitmask.