frangoteam/FUXA #1664: [BUG] View stops updating tag values and freezes in client

Kiruthick-Roxx, 2025-03-25:
**Describe the bug**
Created  view stops updating the tag values to controls  in realtime and freezes when the same view is kept open in a client for some time and requires page reload to restart updating new values

**To Reproduce**
Steps to reproduce the behavior:
1. Create a view with output or guage control
2. Leave the view open for some time without changing the view or page reload
3. Notice the control freezes updating new values and shows the old value 

**Expected behavior**
View and Controls should not stop updating new value and go into sleep after some time irrespective if how much ever time it is open in a browser. Also it should not reload or automatically move to mainview after some time

**Screenshots**

https://github.com/user-attachments/assets/eea3a0ba-af94-4deb-a2d1-0c3eb6c0b9be

**Environment**
 - FUXA version: 1.2.3-2105
 - Node.js version: v18.20.4
 - npm version: 10.9.0
 - Platform/OS: Ubuntu 20.04
 - Browser: Chrome

unocelli, 2025-03-26:
Hi, please try the new version available in master branch that I just published

Kiruthick-Roxx, 2025-03-28:
Hi @unocelli,
I tested it for 2 days, still the control fails to update the value after some time.

I have 2 FUXA clients open in my browser of different FUXA servers.

when I keep one client open for some time and switch to the next one, I dont see the values getting updated. I have to reload the client to make it work.

Im using the latest FUXA master version for both my servers.

unocelli, 2025-03-28:
Hi, what version are you using? the fix is in 1.2.5-2245

Kiruthick-Roxx, 2025-03-28:
Hi @unocelli ,

Im also using 1.2.5-2245 version.

The latest one with that tag subscription fix, but still the problem persists.

henjoe, 2025-03-29:
This is weird, I haven't encountered issue before with websocket. 
@unocelli  
maybe you can try to add reconnect: true to the io client.
```
this.socket = io(`${this.endPointConfig}/?token=${token}`{
reconnect: true,
reconnectionAttempts: 5, // Retry forever (or set a specific number)
reconnectionDelay: 5000, // Initial delay between reconnection attempts (in ms)
reconnectionDelayMax: 60000, // Maximum delay between reconnection attempts (in ms)
});
```

henjoe, 2025-03-29:
Hi @Kiruthick-Roxx I wonder what is the polling rate of your device?

unocelli, 2025-03-29:
Hi, the socket reconnection should work, you have to investigate the socket messages with browser devtools

[image]

Kiruthick-Roxx, 2025-03-29:
> Hi [@Kiruthick-Roxx](https://github.com/Kiruthick-Roxx) I wonder what is the polling rate of your device?

Hi @henjoe,

I'm using OPCUA connected with 25 devices, with 200ms polling rate

Kiruthick-Roxx, 2025-04-02:
Hi @unocelli, @henjoe,
 
The fix works fine, i monitored it for few days and it works properly.
I think we can close this issue.

Thanks for your help.

henjoe, 2025-04-02:
> Hi @unocelli, @henjoe,
>  
> The fix works fine, i monitored it for few days and it works properly.
> I think we can close this issue.
> 
> Thanks for your help.

Hi @Kiruthick-Roxx its good to hear this. To clarify, You mean the fix created by @unocelli right?

Kiruthick-Roxx, 2025-04-02:
Hi @henjoe,

> Hi [@Kiruthick-Roxx](https://github.com/Kiruthick-Roxx) its good to hear this. To clarify, You mean the fix created by [@unocelli](https://github.com/unocelli) right?

Exactly, the fix made by @unocelli.

closed 2025-04-05 by Kiruthick-Roxx