frangoteam/FUXA #1678: JSON payload suscribe topic parsing error ...

Virulazo, 2025-04-01:
Hi, I try to suscribe to a topic with json payload, i select the topic in Broker Topics to Suscribe, then select the topic, and select JSON button, then click on add attribute, but the json is not parsed and the topic is suscribed as raw. How can solve or workaround?  Thanks.

henjoe, 2025-04-04:
Can you give your JSON data format.

Virulazo, 2025-04-04:
Here it is:
{ "nivel":"823.01",
  "caudal":"0.00",
  "litros_ultima_hora":"0.010",
  "bomba":"off",
    "voltage":"217.8",
    "corriente":"0.0",
    "potencia":"0.0",
}

henjoe, 2025-04-04:
> Here it is:
> { "nivel":"823.01",
>   "caudal":"0.00",
>   "litros_ultima_hora":"0.010",
>   "bomba":"off",
>     "voltage":"217.8",
>     "corriente":"0.0",
>     "potencia":"0.0",
> }

The format seems okay. You don't need to click on add attribute to payload, all tag should be automatically mapped when you clicked the JSON.