frangoteam/FUXA #1684: [BUG]After downloading the code, I encountered an issue where 'sqlite3' could not be installed when executing 'npm install' on FUXA master/server

yushenjinghao, 2025-04-10:
**Describe the bug**

**To Reproduce**
Steps to reproduce the behavior:
1. cd ./server
2. npm install
3. An error occurred

**Expected behavior**
Note: FUXA-master/client can run normally, and the problem only occurs when FUXA-master/server is executed.
Basic environment: node: ver18.10.0、npm: ver8.19.2、python-3.10.0-amd64.
After downloading the code, I encountered an issue where 'sqlite3' could not be installed when executing 'npm install' on FUXA master/server.
I have tried using 'npm config' to update the image and downloading 'sqlite3' separately and placing it in 'Node.js', but it was useless.
The following are the errors after execution.
<Error description start>
PS D:\worksoftware\VsCode\download-other\FUXA\FUXA-master\server> npm install
npm WARN deprecated @types/minipass@3.3.5: This is a stub types definition. minipass provides its own type definitions, so you do not need this installed.
npm WARN deprecated @npmcli/move-file@1.1.2: This functionality has been moved to @npmcli/fs
npm WARN deprecated inflight@1.0.6: This module is not supported, and leaks memory. Do not use it. Check out lru-cache if you want a good and tested way to coalesce async requests by a key value, which is much more comprehensive and powerful.
npm WARN deprecated lodash.get@4.4.2: This package is deprecated. Use the optional chaining (?.) operator instead.
npm WARN deprecated npmlog@5.0.1: This package is no longer supported.
npm WARN deprecated npmlog@6.0.2: This package is no longer supported.
npm WARN deprecated rimraf@3.0.2: Rimraf versions prior to v4 are no longer supported
npm WARN deprecated glob@8.1.0: Glob versions prior to v9 are no longer supported
npm WARN deprecated glob@7.2.3: Glob versions prior to v9 are no longer supported
npm WARN deprecated glob@7.2.3: Glob versions prior to v9 are no longer supported
npm WARN deprecated glob@7.2.3: Glob versions prior to v9 are no longer supported
npm WARN deprecated glob@7.2.3: Glob versions prior to v9 are no longer supported
npm WARN deprecated are-we-there-yet@2.0.0: This package is no longer supported.
npm WARN deprecated are-we-there-yet@3.0.1: This package is no longer supported.
npm WARN deprecated debug@4.1.1: Debug versions >=3.2.0 <3.2.7 || >=4 <4.3.1 have a low-severity ReDos regression when used in a Node.js environment. It is recommended you upgrade to 3.2.7 or 4.3.1. (https://github.com/visionmedia/debug/issues/797)    
npm WARN deprecated debug@4.1.1: Debug versions >=3.2.0 <3.2.7 || >=4 <4.3.1 have a low-severity ReDos regression when used in a Node.js environment. It is recommended you upgrade to 3.2.7 or 4.3.1. (https://github.com/visionmedia/debug/issues/797)    
npm WARN deprecated debug@4.1.1: Debug versions >=3.2.0 <3.2.7 || >=4 <4.3.1 have a low-severity ReDos regression when used in a Node.js environment. It is recommended you upgrade to 3.2.7 or 4.3.1. (https://github.com/visionmedia/debug/issues/797)    
npm WARN deprecated gauge@4.0.4: This package is no longer supported.
npm WARN deprecated gauge@3.0.2: This package is no longer supported.
npm WARN deprecated @tdengine/rest@3.0.2: Package no longer supported. Contact Support at https://www.npmjs.com/support for more info.
npm ERR! code 1
npm ERR! path D:\worksoftware\VsCode\download-other\FUXA\FUXA-master\server\node_modules\sqlite3
npm ERR! command failed
npm ERR! command C:\Windows\system32\cmd.exe /d /s /c node-pre-gyp install --fallback-to-build
npm ERR! Failed to execute 'D:\worksoftware\Node\node.exe D:\worksoftware\Nvm\v18.10.0\node_modules\npm\node_modules\node-gyp\bin\node-gyp.js configure --fallback-to-build --module=D:\worksoftware\VsCode\download-other\FUXA\FUXA-master\server\node_modules\sqlite3\lib\binding\napi-v6-win32-unknown-x64\node_sqlite3.node --module_name=node_sqlite3 --module_path=D:\worksoftware\VsCode\download-other\FUXA\FUXA-master\server\node_modules\sqlite3\lib\binding\napi-v6-win32-unknown-x64 --napi_version=8 --node_abi_napi=napi --napi_build_version=6 --node_napi_label=napi-v6' (1)
npm ERR! node-pre-gyp info it worked if it ends with ok
npm ERR! node-pre-gyp info using node-pre-gyp@1.0.11
npm ERR! node-pre-gyp info using node@18.10.0 | win32 | x64
npm ERR! node-pre-gyp info check checked for "D:\worksoftware\VsCode\download-other\FUXA\FUXA-master\server\node_modules\sqlite3\lib\binding\napi-v6-win32-unknown-x64\node_sqlite3.node" (not found)
npm ERR! node-pre-gyp http GET https://npm.taobao.org/mirrors/v5.1.5/napi-v6-win32-unknown-x64.tar.gz
npm ERR! node-pre-gyp ERR! install request to https://npm.taobao.org/mirrors/v5.1.5/napi-v6-win32-unknown-x64.tar.gz failed, reason: Hostname/IP does not match certificate's altnames: Host: npm.taobao.org. is not in the cert's altnames: DNS:*.tbcdn.cn, DNS:*.1688.com, DNS:*.3c.tmall.com, DNS:*.alibaba.com, DNS:*.alicdn.com, DNS:*.aliexpress.com, DNS:*.alikunlun.com, DNS:*.aliqin.tmall.com, DNS:*.alitrip.com, DNS:*.aliyun.com, DNS:*.cainiao.com, DNS:*.cainiao.com.cn, DNS:*.chi.taobao.com, DNS:*.chi.tmall.com, DNS:*.china.taobao.com, DNS:*.cloudvideocdn.taobao.com, DNS:*.cmos.greencompute.org, DNS:*.dingtalk.com, DNS:*.django.t.taobao.com, DNS:*.etao.com, DNS:*.feizhu.cn, DNS:*.feizhu.com, DNS:*.fliggy.com, DNS:*.fliggy.hk, DNS:*.food.tmall.com, DNS:*.jia.taobao.com, DNS:*.jia.tmall.com, DNS:*.ju.taobao.com, DNS:*.juhuasuan.com, DNS:*.lw.aliimg.com, DNS:*.m.1688.com, DNS:*.m.alibaba.com, DNS:*.m.alitrip.com, DNS:*.m.cainiao.com, DNS:*.m.etao.com, DNS:*.m.taobao.com, DNS:*.m.taopiaopiao.com, DNS:*.m.tmall.com, DNS:*.m.tmall.hk, DNS:*.mei.com, DNS:*.mobgslb.tbcache.com, DNS:*.taobao.com, DNS:*.taopiaopiao.com, DNS:*.tbcache.com, DNS:*.tmall.com, DNS:*.tmall.hk, DNS:*.trip.taobao.com, DNS:*.xiami.com, DNS:1688.com, DNS:alibaba.com, DNS:alicdn.com, DNS:aliexpress.com, DNS:alikunlun.com, DNS:alitrip.com, DNS:aliyun.com, DNS:cainiao.com, DNS:cainiao.com.cn, DNS:cloudvideocdn.taobao.com, DNS:cmos.greencompute.org, DNS:dingtalk.com, DNS:etao.com, DNS:feizhu.cn, DNS:feizhu.com, DNS:fliggy.com, DNS:fliggy.hk, DNS:juhuasuan.com, DNS:m.intl.taobao.com, DNS:mei.com, DNS:taobao.com, DNS:taopiaopiao.com, DNS:tmall.com, DNS:tmall.hk, DNS:xiami.com, DNS:tbcdn.cn
npm ERR! node-pre-gyp WARN Pre-built binaries not installable for sqlite3@5.1.5 and node@18.10.0 (node-v108 ABI, unknown) (falling back to source compile with node-gyp)
npm ERR! node-pre-gyp WARN Hit error request to https://npm.taobao.org/mirrors/v5.1.5/napi-v6-win32-unknown-x64.tar.gz failed, reason: Hostname/IP does not match certificate's altnames: Host: npm.taobao.org. is not in the cert's altnames: DNS:*.tbcdn.cn, DNS:*.1688.com, DNS:*.3c.tmall.com, DNS:*.alibaba.com, DNS:*.alicdn.com, DNS:*.aliexpress.com, DNS:*.alikunlun.com, DNS:*.aliqin.tmall.com, DNS:*.alitrip.com, DNS:*.aliyun.com, DNS:*.cainiao.com, DNS:*.cainiao.com.cn, DNS:*.chi.taobao.com, DNS:*.chi.tmall.com, DNS:*.china.taobao.com, DNS:*.cloudvideocdn.taobao.com, DNS:*.cmos.greencompute.org, DNS:*.dingtalk.com, DNS:*.django.t.taobao.com, DNS:*.etao.com, DNS:*.feizhu.cn, DNS:*.feizhu.com, DNS:*.fliggy.com, DNS:*.fliggy.hk, DNS:*.food.tmall.com, DNS:*.jia.taobao.com, DNS:*.jia.tmall.com, DNS:*.ju.taobao.com, DNS:*.juhuasuan.com, DNS:*.lw.aliimg.com, DNS:*.m.1688.com, DNS:*.m.alibaba.com, DNS:*.m.alitrip.com, DNS:*.m.cainiao.com, DNS:*.m.etao.com, DNS:*.m.taobao.com, DNS:*.m.taopiaopiao.com, DNS:*.m.tmall.com, DNS:*.m.tmall.hk, DNS:*.mei.com, DNS:*.mobgslb.tbcache.com, DNS:*.taobao.com, DNS:*.taopiaopiao.com, DNS:*.tbcache.com, DNS:*.tmall.com, DNS:*.tmall.hk, DNS:*.trip.taobao.com, DNS:*.xiami.com, DNS:1688.com, DNS:alibaba.com, DNS:alicdn.com, DNS:aliexpress.com, DNS:alikunlun.com, DNS:alitrip.com, DNS:aliyun.com, DNS:cainiao.com, DNS:cainiao.com.cn, DNS:cloudvideocdn.taobao.com, DNS:cmos.greencompute.org, DNS:dingtalk.com, DNS:etao.com, DNS:feizhu.cn, DNS:feizhu.com, DNS:fliggy.com, DNS:fliggy.hk, DNS:juhuasuan.com, DNS:m.intl.taobao.com, DNS:mei.com, DNS:taobao.com, DNS:taopiaopiao.com, DNS:tmall.com, DNS:tmall.hk, DNS:xiami.com, DNS:tbcdn.cn
npm ERR! gyp info it worked if it ends with ok
npm ERR! gyp info using node-gyp@9.1.0
npm ERR! gyp info using node@18.10.0 | win32 | x64
npm ERR! gyp info ok
npm ERR! gyp info it worked if it ends with ok
npm ERR! gyp info using node-gyp@9.1.0
npm ERR! gyp info using node@18.10.0 | win32 | x64
npm ERR! gyp info find Python using Python version 3.10.0 found at "D:\worksoftware\Python\python.exe"
npm ERR! gyp ERR! find VS
npm ERR! gyp ERR! find VS msvs_version not set from command line or npm config
[... cut ...]
PS D:\worksoftware\VsCode\download-other\FUXA\FUXA-master\server> nvm ls

    22.0.0
    21.1.0
    20.10.0
    18.18.0
    18.17.0
  * 18.10.0 (Currently using 64-bit executable)
    14.15.1
PS D:\worksoftware\VsCode\download-other\FUXA\FUXA-master\server> npm -v
8.19.2
PS D:\worksoftware\VsCode\download-other\FUXA\FUXA-master\server>
<Error description end>

**Screenshots**

**Example project**

**Environment**
 - FUXA version: 1.2.4
 - Node.js version: 18.10.0
 - npm version: 8.19.2
 - Platform/OS: Windows11
 - Browser: chrome（106.0.5249.62）

jingshui127, 2025-04-15:
一般和你的python版本有关。还有就是环境变量有问题

yushenjinghao, 2025-04-15:
可否说一下推荐的python版本，还有环境变量的配置项，请给个参考示例，我会尝试一下。

yushenjinghao, 2025-04-15:
安装 Visual Studio 的 C++ 编译环境后解决了此问题

unocelli, 2025-05-04:
I’m closing this issue since there is nothing feasible and due to inactivity.

closed 2025-05-04 by unocelli