frangoteam/FUXA #1692: [FEATURE]Access variable historical data through APIs.

jingshui127, 2025-04-15:
**Describe the feature**
Access variable historical data through APIs.

**Describe the solution you'd like**
I want to extend some functionality myself, and I need to be able to access the historical data of variables. How? Is historical data accessible via API?

unocelli, 2025-04-15:
Hi, look the  '/api/daq' endpoint

[FUXA_WebAPI.pdf](https://github.com/user-attachments/files/19762434/FUXA_WebAPI.pdf)

jingshui127, 2025-04-17:
[image]Modifying parameters doesn't seem to work!

[image]

unocelli, 2025-04-17:
Hi, historical data are not to be found in the _images folder