frangoteam/FUXA #1695: [BUG] Weird behaviour with setView

carlbomsdata, 2025-04-16:
I have the following client side script:
```
try{
	const id = $getTagId("test_tag") 
        const val = $getTag(id)

	$setTag(id, val + 1)

	console.log(id)
  
  	if ((val + 1) == 3) {
    	    $setView('mobile');
    }
  
}catch(e){alert(e.message)}
```
On my main view i have a button that triggers this script.

When '$setView('mobile');' gets triggered it changes the view to "mobile" view but it looks like this:
[image]

Also on main view i have a button (click=>open page=>mobile) that opens the mobile view. When i click it i get:
[image]

Is this a known issue?

carlbomsdata, 2025-04-17:
Could you please advise in which files i should look at so i can help fixing the problem.

carlbomsdata, 2025-04-17:
For some reason it seems that when calling $setView from script it also spawns a Card. 
[image]

When setting the css class .fab-card[_ngcontent-cid-c143] to display:none; it now looks correct, although the problem is still there..
[image]

unocelli, 2025-04-17:
Hi, I don't have any issues
[image]

carlbomsdata, 2025-04-18:
You are correct. The problem seems to only be occurring in the latest release (v1.2.4).
Cloning the repository, does not have the problem :)

carlbomsdata, 2025-04-19:
If it is at any help.
The problem seems to be here: 'C:\Users\tobias\Documents\Development\fuxa_responsive\FUXA-1.2.4\client\src\app\_services\hmi.service.ts'

Inside onScriptCommand (572).
Original code
```
    public onScriptCommand(message: ScriptCommandMessage) {
        if (message.params && message.params.length) {
            switch (message.command) {
                case ScriptCommandEnum.SETVIEW:
                    this.onGoTo.emit(<ScriptSetView>{ viewName: message.params[0], force: message.params[1] });
				case ScriptCommandEnum.OPENCARD:
                    this.onOpen.emit(<ScriptOpenCard>{ viewName: message.params[0], options: message.params[1] });
				break;
            }
        }
    }
```

Debug code
```
    public onScriptCommand(message: ScriptCommandMessage) {
        if (message.params && message.params.length) {
            switch (message.command) {
                case ScriptCommandEnum.SETVIEW:
                    this.onGoTo.emit(<ScriptSetView>{ viewName: message.params[0], force: message.params[1] });
					console.log("setview");
				case ScriptCommandEnum.OPENCARD:
                    this.onOpen.emit(<ScriptOpenCard>{ viewName: message.params[0], options: message.params[1] });
					console.log("opencard");
				break;
            }
        }
    }
```
[image]

Solution
```
public onScriptCommand(message: ScriptCommandMessage) {
        if (message.params && message.params.length) {
            switch (message.command) {
                case ScriptCommandEnum.SETVIEW:
                    this.onGoTo.emit(<ScriptSetView>{ viewName: message.params[0], force: message.params[1] });
					break;
				case ScriptCommandEnum.OPENCARD:
                    this.onOpen.emit(<ScriptOpenCard>{ viewName: message.params[0], options: message.params[1] });
					break;
				default:
					break;
            }
        }
    }
```

unocelli, 2025-04-20:
Yes right, I also realised, I fixed it but not yet pushed, now I do, thank you

closed 2025-04-20 by carlbomsdata

rdauanny, 2025-06-12:
Hello guys! Is this solved? I am still facing the $setView issue. Thanks in advance.

rdauanny, 2025-06-13:
It is working! I've just upated docker image! Thanks!