frangoteam/FUXA #1696: [BUG] Open view from multiview

carlbomsdata, 2025-04-16:
My start page is a multi view which displays a view called View1.
Inside View1 there is a button with click=>open view=View2.

Inside View2 there is a button with click=>open view=View1.

When clicking on the button inside View1 then View2 gets somewhat opened inside the multi view. I expect View2 to only get shown and  multi view to close.

When clicking on the button inside View2, the View1 also opens inside the multi view but I only see the background color of View1.

unocelli, 2025-04-17:
Hi, you mean that the background of the multi view child should adjust to the view shown?

henjoe, 2025-05-08:
This is what he meant @unocelli which kinda strange.

[image]

unocelli, 2026-11-11:
Hi @tobias-carlbom I can reproduce it, thanks for the report. In any case i think you can achieve your purpose using scripts ($setView)