frangoteam/FUXA #1703: How to configure address for a 20-character string in Siemens S7-1500

fengchenyu, 2025-04-22:
[image]
Hi,

Hi  FUXA author,

I am working with Siemens S7-1500, and I need to configure an address for a string variable. The variable is defined as:

Variable Name: Query_CODE

Type: Array[0..19] of Char

Offset: 4~23

Data Block: DB1

Could you kindly provide guidance on how to configure the address for this 20-character string in the context of S7-1500?

Thank you for your help!

Best regards

unocelli, 2025-04-27:
Hi, the feature is missing #692

fengchenyu, 2025-04-28:
There is also another issue: after packaging FUXA with Electron, the S7 communication fails because node-snap7 is a native module. Although it was correctly unpacked into the app.asar.unpacked directory, the application code does not correctly resolve the path to the native module after packaging. As a result, the module fails to load at runtime, causing connection issues with the S7 PLCs.
﻿
Could you please advise on the correct way to handle this issue?